WOA-Project/WOA-Deployer-Rpi #5: Trojan found in WOA.Deployer.Console.zip

margalex, 2019-02-19:
Unzipping the WOA.Deployer.Console.zip makes Windows Defender detects this trojan:

https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fEmali.A!cl&threatid=2147728322

CornyjK, 2019-02-19:
I think we can do nothing about this, just whitelist it.

margalex, 2019-02-19:
Windows Defender reports that WoaDeployer.exe is infected in WOA.Deployer.Console.zip. I didn't get any warning with WOA.Deployer.GUI.zip.

CornyjK, 2019-02-19:
Yes, we can do nothing about this currently, just whitelist it.

SuperJMN, 2019-02-19:
I think somebody tagged the download as malware... all of a sudden, I can't download the .zip!

[image]

I've already sent a removal request to the Windows Defender team:
[image]

By the way, this is the virustotal scan summary: https://www.virustotal.com/en/file/1bfbe1372b9afcec086645d24f97fe726d7093c1bcfbfec89329fff754b53d97/analysis/1550613259/

Only Windows Defender is reporting it as malware 😒

SuperJMN, 2019-02-20:
@margalex Can you, please, try again? The Windows Defender team seems to have removed the false positive from it.

They gave me this instructions to "clear" the cached results:

> We have removed the detection. Please follow the steps below to clear cached detection and obtain the latest malware definitions.
> 
> 1. Open command prompt as administrator and change directory to c:\Program Files\Windows Defender 
> 2. Run “MpCmdRun.exe -removedefinitions -dynamicsignatures”

margalex, 2019-02-20:
@SuperJMN, I tried it again and it's OK now. Thx!

closed 2019-02-20 by SuperJMN

CornyjK, 2019-02-20:
> Can you, please, try again? The Windows Defender team seems to have removed the false positive from it.

I expected MS to be much slower 😀