WOA-Project/WOA-Deployer-Rpi #6: Could not read the metadata from the WIM file

geeksfn, 2019-02-20:
I am getting an error when selecting the install.wim file, I have tried downloading the ISO again however its resulting in the same error.

The ISO is 'Windows 10, Version 1809 (Updated Nov '18) ARM 64 Bit English'

2019-02-20 09:40:43.050 +03:00 [Verbose] Default directory for WimFolder has been set to "D:\Win10OnPi\1809_ARM\sources"
2019-02-20 09:40:43.055 +03:00 [Verbose] Trying to load WIM metadata file at '"D:\Win10OnPi\1809_ARM\sources\install.wim"'
2019-02-20 09:40:43.077 +03:00 [Verbose] Getting WIM stream
2019-02-20 09:40:43.078 +03:00 [Verbose] (WIM) Finding Magic Bytes...
2019-02-20 09:40:43.079 +03:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-02-20 09:40:43.079 +03:00 [Verbose] (WIM) Finding WIM XML Data...
2019-02-20 09:40:43.079 +03:00 [Verbose] (WIM) Found WIM XML Data at 035639643382
2019-02-20 09:40:43.144 +03:00 [Error] WIM file error
Deployer.Exceptions.InvalidWimFileException: Could not read the metadata from the WIM file. Please, check it's a valid .WIM file ---> System.InvalidOperationException: There is an error in XML document (1, 5147). ---> System.Xml.XmlException: Unexpected end of file has occurred. The following elements are not closed: WIM. Line 1, position 5147.
   at System.Xml.XmlTextReaderImpl.Throw(Exception e)
   at System.Xml.XmlTextReaderImpl.Throw(String res, String arg)
   at System.Xml.XmlTextReaderImpl.ThrowUnclosedElements()
   at System.Xml.XmlTextReaderImpl.ParseElementContent()
   at System.Xml.XmlTextReaderImpl.Read()
   at System.Xml.XmlTextReader.Read()
   at System.Xml.XmlReader.ReadEndElement()
   at System.Xml.Serialization.XmlSerializationReader.ReadEndElement()
   at Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderWimMetadata.Read7_ImageMetadata(Boolean isNullable, Boolean checkType)
   at Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderWimMetadata.Read8_WimMetadata(Boolean isNullable, Boolean checkType)
   at Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderWimMetadata.Read9_WIM()
   --- End of inner exception stack trace ---
   at System.Xml.Serialization.XmlSerializer.Deserialize(XmlReader xmlReader, String encodingStyle, XmlDeserializationEvents events)
   at System.Xml.Serialization.XmlSerializer.Deserialize(Stream stream)
   at Deployer.Services.Wim.WindowsImageMetadataReaderBase.Load(Stream stream)
   --- End of inner exception stack trace ---
   at Deployer.Services.Wim.WindowsImageMetadataReaderBase.Load(Stream stream)
   at Deployer.Raspberry.Gui.ViewModels.WimPickViewModel.LoadWimMetadata(String path) in D:\a\1\s\Source\Deployer.Raspberry.Gui\ViewModels\WimPickViewModel.cs:line 87
   at System.Reactive.Linq.ObservableImpl.Select`2.Selector._.OnNext(TSource value)

SuperJMN, 2019-03-14:
Unfortunately, .wim files for some versions aren't valid. It's not a problem of WOA Deployer, but the .wim files themselves. Try with another version. For instance: 18850.1000.190301-2054.RS_PRERELEASE.

closed 2019-03-14 by SuperJMN

AndreasStokidis, 2019-04-02:
I had the same issue with five different version of windows. I updated windows to last version (October) and the card has been flashed correctly.