WOA-Project/WOA-Deployer-Rpi #7: Problems with the first boot

Raideerke, 2019-02-20:
Because the old repo was archived, i open a new one here.
So created an SD card with 1809 Windows and core package which has been provided and after probably a hour of initialise progress, the system crashed before the frist boot.

I was assuming that will be in the section where i can choose user, and some privacy info .. but it crashed.
I'm using a hungarian version of windows, but the error code is still in international format :)

[image]

If you have any ideas what i should do to fix the issue, i would be happy.

Raideerke, 2019-02-20:
Will try to flash the system again to an another drive (probably will not fix the issue), waiting for an SSD and a SATA->USB adapter with external power.

SuperJMN, 2019-02-20:
Please, try 17134 to see if it works.

Obtener Outlook para Android<https://aka.ms/ghei36>

Raideerke, 2019-02-21:
Thank you! Yes. With 1803 (17134 build) it works well.
I see that the CPU is downclocked to 0,6 while using windows. I have cooling system on the pi. It is possible to use the CPU on the maximum clock speeds?

capriciousduck, 2019-02-21:
You could do it in the bios. Search for "Chipset Configuration" and there you will find the option and set it to max.

closed 2019-02-21 by CornyjK