WOA-Project/WOA-Deployer-Rpi #10: Keyboard not detected during windows setup

Kenji94, 2019-02-21:
Hi,

First thank your for your works.
My keyboard is detected when i'm on the bios but not during windows setup.
I'm stuck at the language screen selection.

Do you have a solution for this ?

paerra, 2019-02-21:
Hi  I have the same problem. Keyboard works fine in bios. 
When the step "choose keyboardlanguage" the usb ports stops working. Tried booth bluetooth and wired keyboard.

osakanataro, 2019-02-22:
In my environment, same problem  has occurred.
Windows 10 1809 could not work. 
Bluescreen and crash many times.
Windows 10 1803 or Insider Preview 18836 work fine.
Notthing bluescreen and crash.

amankumarsingh32, 2019-02-22:
keyboard not working after booting. If both keyboard and mouse are connected bluescreen comes and crashes. without keyboard and mouse goes to choose keyboard language. In BIOS working fine.

paerra, 2019-02-22:
Hi again

Tried version Insider Preview 18836 and it goes fine until account setting. (Tried booth offline and connected with rj45). Errorcode
”OOBESETTINGSMULTIPAGE” and try again. Can’t skip this step

paerra, 2019-02-22:
Here is a link to a picture on the error code. 
Translated to ”an error occured”

https://ibb.co/mcQ8sbR

paerra, 2019-02-22:
Finally! Choose to use my existing Microsoft account and password. 
Up and running

CornyjK, 2019-02-23:
If you need further help, read [this](https://github.com/WOA-Project/WOA-Deployer-Rpi/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

closed 2019-02-23 by CornyjK