WOA-Project/WOA-Deployer-Rpi #11: Error:sequence contains no elements

captaindannduece, 2019-02-23:
Right after writing and checking the GPT table.

SuperJMN, 2019-02-23:
Can you try another card reader? Just to exclude a failing one. Or another PC. Thanks!

captaindannduece, 2019-02-23:
Still same problem.

CornyjK, 2019-02-23:
Which readers did you try? Do both PC you used use Windows 10?

captaindannduece, 2019-02-23:
I used both win 10 computers, I used both one's sd card reader in an adapter and a USB device that has a micro sd slot.

captaindannduece, 2019-02-23:
Now

Error:Object reference not set to an instance of an object.

CornyjK, 2019-02-24:
Do you have the log file?

captaindannduece, 2019-02-25:
Closing because no longer original issue.

captaindannduece, 2019-02-26:
2019-02-26 13:48:51.560 -08:00 [Verbose] The '"C:\Users\Daniel\Desktop"' exists
2019-02-26 13:48:56.429 -08:00 [Verbose] Default directory for WimFolder has been set to "C:\Users\Daniel\Desktop"
2019-02-26 13:48:56.436 -08:00 [Verbose] Trying to load WIM metadata file at '"C:\Users\Daniel\Desktop\install.wim"'
2019-02-26 13:48:56.515 -08:00 [Verbose] Getting WIM stream
2019-02-26 13:48:56.517 -08:00 [Verbose] (WIM) Finding Magic Bytes...
2019-02-26 13:48:56.518 -08:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-02-26 13:48:56.518 -08:00 [Verbose] (WIM) Finding WIM XML Data...
2019-02-26 13:48:56.519 -08:00 [Verbose] (WIM) Found WIM XML Data at 043405578422
2019-02-26 13:48:56.564 -08:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4340557842", Images: [ImageMetadata { DirectoryCount: "22458", FileCount: "97876", TotalBytes: "16964993283", HardLinkBytes: "6518216411", CreationTime: Time { HighPart: "0x01D44CEB", LowPart: "0xF4315121" }, LastModificationTime: Time { HighPart: "0x01D4C75D", LowPart: "0x21F3FC72" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Microsoft® Windows® Operating System", EditionId: "Core", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.17763.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "en-US", Default: "en-US" }, Version: Version { Major: "10", Minor: "0", Build: "17763", SpBuild: "292", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Home", Description: "Windows 10 Home", Flags: null, DiplayName: "Windows 10 Home", DisplayDescription: "Windows 10 Home", Index: "1" }] }
2019-02-26 13:48:56.571 -08:00 [Verbose] WIM metadata file at '"C:\Users\Daniel\Desktop\install.wim"' retrieved correctly
2019-02-26 13:49:07.189 -08:00 [Information] Downloading UEFI
2019-02-26 13:49:07.234 -08:00 [Warning] UEFI was already downloaded. Skipping download.
2019-02-26 13:49:07.242 -08:00 [Information] Zip Unpack from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan95xx-windows/lan9500-wdf-v18.12.18.0.zip to Downloaded\Drivers\LAN95XX
2019-02-26 13:49:07.256 -08:00 [Information] Zip Unpack from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan78xx-windows/lan7800-wdf-v18.12.14.0.zip to Downloaded\Drivers\LAN78XX
2019-02-26 13:49:07.265 -08:00 [Information] Zip Unpack from https://pi64.win/wp-content/uploads/2019/02/usb-drivers.zip to Downloaded\Drivers\USB
2019-02-26 13:49:07.270 -08:00 [Information] Fetching from GitHub subfolder: https://github.com/driver1998/bsp/bsp-master/prebuilt to Downloaded\Drivers\BSP Drivers
2019-02-26 13:49:07.272 -08:00 [Warning] "https://github.com/driver1998/bsp""bsp-master/prebuilt" was already downloaded. Skipping download.
2019-02-26 13:49:07.280 -08:00 [Information] License from Downloaded\Drivers\USB\license.md
2019-02-26 13:49:10.585 -08:00 [Information] Deploying Windows
2019-02-26 13:49:10.589 -08:00 [Information] Flashing GPT image...
2019-02-26 13:49:10.597 -08:00 [Verbose] We are about to run Etcher: "Core\Tools\Etcher-Cli\x86\Etcher" "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount"
2019-02-26 13:49:10.602 -08:00 [Verbose] Starting process { FileName: "Core\Tools\Etcher-Cli\x86\Etcher", Arguments: "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount" }
2019-02-26 13:49:10.915 -08:00 [Verbose] Process started sucessfully
2019-02-26 13:50:08.592 -08:00 [Information] Validating flashed image...
2019-02-26 13:50:59.663 -08:00 [Information] GPT image flashed
2019-02-26 13:50:59.668 -08:00 [Verbose] Creating Windows partition...
2019-02-26 13:51:02.297 -08:00 [Verbose] Mounting volume "Label: , Size: 0 b, Partition: Disk: Number: 1, Size: 59.48 GB, AllocatedSize: 1 MB, Number: 3, Letter: "
2019-02-26 13:51:36.933 -08:00 [Verbose] Windows Partition created successfully
2019-02-26 13:51:36.936 -08:00 [Information] Preparing for Windows deployment
2019-02-26 13:51:36.953 -08:00 [Verbose] Getting "WindowsARM" volume
2019-02-26 13:51:38.923 -08:00 [Information] Deploying Windows...
2019-02-26 13:51:38.931 -08:00 [Verbose] Checking image at "C:\Users\Daniel\Desktop\install.wim", with index "C:\Users\Daniel\Desktop\install.wim"
2019-02-26 13:51:38.932 -08:00 [Verbose] Image file at '"C:\Users\Daniel\Desktop\install.wim"' exists
2019-02-26 13:51:38.932 -08:00 [Verbose] Getting WIM stream
2019-02-26 13:51:38.932 -08:00 [Verbose] (WIM) Finding Magic Bytes...
2019-02-26 13:51:38.932 -08:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-02-26 13:51:38.932 -08:00 [Verbose] (WIM) Finding WIM XML Data...
2019-02-26 13:51:38.932 -08:00 [Verbose] (WIM) Found WIM XML Data at 043405578422
2019-02-26 13:51:38.949 -08:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4340557842", Images: [ImageMetadata { DirectoryCount: "22458", FileCount: "97876", TotalBytes: "16964993283", HardLinkBytes: "6518216411", CreationTime: Time { HighPart: "0x01D44CEB", LowPart: "0xF4315121" }, LastModificationTime: Time { HighPart: "0x01D4C75D", LowPart: "0x21F3FC72" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Microsoft® Windows® Operating System", EditionId: "Core", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.17763.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "en-US", Default: "en-US" }, Version: Version { Major [... cut ...]
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.Gui.ViewModels.DeploymentViewModel.<Deploy>d__20.MoveNext() in D:\a\1\s\Source\Deployer.Raspberry.Gui\ViewModels\DeploymentViewModel.cs:line 82
2019-02-26 13:54:48.999 -08:00 [Information] Error: There has been a problem during deployment: DISM exited with code 3.

CornyjK, 2019-02-26:
What size does your SD card have?

captaindannduece, 2019-02-26:
64 gb

captaindannduece, 2019-03-07:
[Log-20190306.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/2939562/Log-20190306.txt)
update 

Error: Value cannot be null.
Parameter name: volume

SuperJMN, 2019-03-07:
@captaindannduece I would need to debug this. Do you have TeamViewer? Please, contact me in Telegram using my username "SuperJMN". Thanks.

captaindannduece, 2019-03-14:
Keyboard not responding after Windows boots.

SuperJMN, 2019-03-14:
@captaindannduece sorry, but that isn't the kind of issue to be reported here. It has nothing to do with WOA Deployer. WOA Deployer isn't WOA, nor is DISM.

closed 2019-03-14 by SuperJMN