WOA-Project/WOA-Deployer-Rpi #12: DISM Exited With Code 112

patel8786, 2019-02-23:
Same problem reported in WOA Installer

2019-02-23 08:58:49.885 +02:00 [Verbose] (WIM) Found WIM XML Data at 032047808422
2019-02-23 08:58:49.904 +02:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "3204780842", Images: [ImageMetadata { DirectoryCount: "19016", FileCount: "85770", TotalBytes: "13896743737", HardLinkBytes: "6092415839", CreationTime: Time { HighPart: "0x01D4C09E", LowPart: "0xE35BD26F" }, LastModificationTime: Time { HighPart: "0x01D4C09F", LowPart: "0xA17D02E1" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Microsoft® Windows® Operating System", EditionId: "Core", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.18836.1000;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "en-GB", Default: "en-GB" }, Version: Version { Major: "10", Minor: "0", Build: "18836", SpBuild: "1000", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Home", Description: "Windows 10 Home", Flags: null, DiplayName: "Windows 10 Home", DisplayDescription: "Windows 10 Home", Index: "1" }] }
2019-02-23 08:58:49.905 +02:00 [Verbose] We are about to run DISM: "C:\WINDOWS\SysNative\dism.exe" "/Apply-Image  /ImageFile:\"C:\Users\Mo Patel\Downloads\WoA\install.wim\" /Index:1 /ApplyDir:E:\"
2019-02-23 08:58:49.905 +02:00 [Verbose] Starting process { FileName: "C:\WINDOWS\SysNative\dism.exe", Arguments: "/Apply-Image  /ImageFile:\"C:\Users\Mo Patel\Downloads\WoA\install.wim\" /Index:1 /ApplyDir:E:\" }
2019-02-23 08:58:49.950 +02:00 [Verbose] Process started sucessfully
2019-02-23 09:41:26.014 +02:00 [Error] An error has occurred
Deployer.Exceptions.DeploymentException: There has been a problem during deployment: DISM exited with code 112.
   at Deployer.Filesystem.FullFx.DismImageService.<ApplyImage>d__1.MoveNext() in D:\a\1\s\Source\DeployerPlatform\Deployer.NetFx\DismImageService.cs:line 46
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.WindowsDeployer.<Deploy>d__7.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Tasks.DeployWindows.<Execute>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.WoaDeployer.<Deploy>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.Gui.ViewModels.DeploymentViewModel.<Deploy>d__20.MoveNext() in D:\a\1\s\Source\Deployer.Raspberry.Gui\ViewModels\DeploymentViewModel.cs:line 82
2019-02-23 09:41:26.056 +02:00 [Information] Error: There has been a problem during deployment: DISM exited with code 112.

SuperJMN, 2019-02-23:
I've seen 112 means that there's not enough space on the disk. What size is your SD Card?

patel8786, 2019-02-23:
8GB (showing as 7.4GB)

SuperJMN, 2019-02-23:
Sorry, but that's not enough for Windows. Get a bigger one, please 😅

closed 2019-02-23 by SuperJMN

Rajivp87, 2019-03-12:
how can solve error

System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)    at Deployer.Raspberry.Gui.ViewModels.DeploymentViewModel.<Deploy>d__20.MoveNext() in D:\a\1\s\Source\Deployer.Raspberry.Gui\ViewModels\DeploymentViewModel.cs:line 82