WOA-Project/WOA-Deployer-Rpi #23: Clean temp files

CorentinDy, 2019-03-07:
Hi, 

is there any way to clean the 9566/temp folder created after the build ? 
Around 15,6Go is a pretty big usage of my disk and it seems that I don't have permissions to delete all the files.

Thx

SuperJMN, 2019-03-07:
Hello! What folder are you talking about? Remember that WOA Deployer only DEPLOYS Windows.

BlakeAvery, 2019-03-09:
I have this issue as well, it's connected to the batch file used to build the ARM64 ISO. If you exit the batch file before it's done building the ISO, the temp files the batch script grabs can't be removed. I even tried booting into a Linux flash drive to delete the files, and it didn't work.

SuperJMN, 2019-03-09:
I'm sorry, but that's not an issue of WOA Deployer. Have you tried to generate the iso files using http://uupdump.ml?

closed 2019-03-09 by SuperJMN

CorentinDy, 2019-03-11:
I think it's temporary files from this procedure -> https://github.com/WOA-Project/guides/blob/master/GettingWOA.md