WOA-Project/WOA-Deployer-Rpi #26: rock64/pine64/rockpro64 support.

necrose99, 2019-03-16:
**http://wiki.pine64.org/index.php/ROCKPro64_Software_Release

http://files.pine64.org/doc/rock64/tools/DriverAssitant_v4.5.zip  windows drivers.** 

release for rock64/rockpro64 is not up..  pine64 image however is.. and has remained stable. 

however the boards are similar to RPI3  however 4 gigs of ram , NVME/m.2 support on rockpro64 or sata via PCIE-x4 might be of use..    (IE media server) 

 PINE64 Win10 IoT   needs boot files replaced..  ,  for them to work , 

the **powershell/scripts or other  build your own image** from **<u>WOA-Deployer-Rpi</u>** to bring iot to a full. 

could be an option,   ie Base IOT ,  then scripts append the image.. 
board i have to test on other than RPI3-A , is the rock64pro , with LDC camera case etc, 
as to try to shove into case a header clip and micro pcie-4 to nvme not the pine stores as its a more full height. 
(Micro Connectors PCIE-M21U80HS low-profile M.2 NVMe SSD to PCIe X4 via amazon B07J6NHX8H)  

as Pine64 has nas cases and cards makes for a w10 powered nas/media server , something most of the house gets  (since linux to the rest of them is 'voodoo')  

[image]

*   PINE64 Win10 IoT build already passed the<span> </span>[Microsoft Azure certification](https://catalog.azureiotsuite.com/details?title=Allwinner_Technology_Pine64)
*   For step by step installation process, please follow this<span> </span>[github link](https://github.com/Leeway213/Win10-IoT-for-A64-Release-Notes/blob/master/doc/How%20to%20flash%20ffu.md)
*   For release note, please follow this<span> </span>[github link](https://github.com/Leeway213/Win10-IoT-for-A64-Release-Notes/blob/master/20160809/Pine64/ReleaseNotes.md)
*   For Microsoft Azure IoT SDKs, please follow this<span> </span>[github link](https://github.com/Azure/azure-iot-sdks/)

CornyjK, 2019-03-16:
This is not deployer of the IoT version

closed 2019-03-16 by CornyjK

necrose99, 2019-03-16:
I merely added the IOT for reference as the base iot build dose exist. 
nor as of yet a WOA build for PINE/ROCK64 devices.. 
(if RPI would add some beef  ... ie RPI4  with 4-8 gis of ram , i'd  likely keep to RPI however some tasks the rams just quite weak.) 

**as a Full WOA-Deployer-pine64 , rock64,rockpro64**  
would be a more useful item than the IOT ...
ie **W10-IOT >>>>**> **FULL WOA install conversion..** via scripts perhaps. 

I was merely asking if because the IOT package is available , if ... perhaps a WOA build could be much more easily added. since a iot build is available.. 
I did add the link to  drivers that were on hand ...

otherwise any IOT to WOA build conversion tools might be a useful item to have. 
IOT in WOA build out..  for community use.. 

upgrading from IOT to WOA ??? and or perhaps adding support to the deployer to do WOA on said SBC's.