WOA-Project/WOA-Deployer-Rpi #28: Error on Deploy: "Could not find a part of the path 'C:\WINDOWS\system32\Scripts\Deployment.txt'"

eximiusvir, 2019-04-06:
Please use the "Preview" tab above to view this message if you are seeing this in the new issue text box.

Please note that this is a **WoA Deployer** issue tracker, use this issue tracker only for the WoA Deployer related bugs.

Please open ticket in [Raspberry Pi WOA forum](https://discourse.pi64.win/) or [Raspberry Pi WOA Telegram group](https://t.me/joinchat/ACWtm01aRdQuRAu1d3kW7Q) for anything else, such as:

- Reporting bugs of other tools, drivers, etc.
- Asking for help

# Please remove the text above before opening an issue.

**Describe the bug**
After following the steps to generate the WOA ISO ([link](https://github.com/WOA-Project/guides/blob/master/GettingWOA.md)) and proceeding to exercise the WOA Deployer for Raspberry PI (2.11), when I click Deploy, I receive the error: "Could not find a part of the path 'C:\WINDOWS\system32\Scripts\Deployment.txt'.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to https://github.com/WOA-Project/guides/blob/master/GettingWOA.md and follow the steps to download an ARM64 ISO installer image.
2. Go to https://github.com/WOA-Project/WOA-Deployer-Rpi and download the latest version ([link](https://github.com/WOA-Project/WOA-Deployer-Rpi/releases/latest))
3. Mount the ISO image (E:\)
3. Extract WOA.Deployer to C:\temp\WOA.Deployer
4. In File Explorer, navigate to C:\temp\WOA.Deployer and double click WoaDeployer.exe
5. In the DRIVE SELECTION drop-down, select your MicroSD card (mine is 64GB)
6. In the WINDOWS IMAGE, specify the path E:\sources\install.wim
7. Note the drop-down below the path will populate with: **1** Windows 10 Pro build 17763
8. Click Deploy
9. Click OK to Confirm deployment
10. Note the displayed error:
Error
Could not find a part of the path 'C:\WINDOWS\system32\Scripts\Deployment.txt'

**Expected behavior**
Deployment would proceed as expected

[image]

CornyjK, 2019-04-10:
Check your deployer/redownload it, your scripts folder is probably missing. If issue persists, comment.

closed 2019-04-10 by CornyjK

vimalswaroopj, 2020-04-11:
run the WOA software as administrator. First I tried many times with out using Run as Administrator option. Later I tried with it and it worked for me.

Cycododge, 2020-05-05:
For myself, the issue was caused by doing a `*.exe` search in the deployer folder, and attempting to run it from the results view. Clear the search and locate it manually.