WOA-Project/WOA-Deployer-Rpi #36: There is a problem with the WOA.DEPLOYER detection of the disks

yoyoyoyoking, 2019-06-01:
There is a problem with the WOA.DEPLOYER detection of the disks
It doesn't detected 2 of the disks that I have
if i do diskpart and then type disk list, it shows the 2 disks
But WOA.DEPLOYER doesnt show both of them...

Geeviz, 2019-06-18:
Here the same problem too. 

At my work there was (after restart pc) no problem but at home the program will not show any of my discs. Maybe this is because at my work I have windows 10 october 2018 and at home may 2019..

Geeviz, 2019-06-19:
Problem 2 is as follows:  program crashes immediately when pointing to the install.wim ... now trying program compatibility to windows xp/7/8/vista

Geeviz, 2019-06-19:
[ A tip: Do a fresh install of Windows 10 may 2019 and all your problems are gone. it will detect your devices and it do not crashes by pointing to the correct install.wim image. ]

Geeviz, 2019-06-19:
Or as less rigorous: try to de-install old and re-install the newest Net Framework..