WOA-Project/WOA-Deployer-Rpi #39: Crashes immediately when pointing to install.wim file, not every comp shows harddisk/usb devices

Geeviz, 2019-06-19:
1. On some computers when loading the program it will not show up any usb (harddisk) device
2. Program crashes immediately after pointing to location of install.wim... it tries to load and *bam* .. crash

frumbert, 2019-06-25:
Wow, you get a crash?? All I get is the spinning wheel of doom.

The log is pretty nondescript

```
2019-06-25 18:46:54.401 +10:00 [Verbose] Default directory for WimFolder has been set to "C:\"
2019-06-25 18:46:54.405 +10:00 [Verbose] Trying to load WIM metadata file at '"C:\install.wim"'
2019-06-25 18:46:54.473 +10:00 [Verbose] Getting WIM stream
2019-06-25 18:46:54.475 +10:00 [Verbose] (WIM) Finding Magic Bytes...
2019-06-25 18:46:54.475 +10:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-06-25 18:46:54.475 +10:00 [Verbose] (WIM) Finding WIM XML Data...
2019-06-25 18:46:54.475 +10:00 [Verbose] (WIM) Found WIM XML Data at 002
2019-06-25 18:49:50.284 +10:00 [Verbose] The '"C:\"' exists
2019-06-25 18:49:54.049 +10:00 [Verbose] Default directory for WimFolder has been set to "C:\"
2019-06-25 18:49:54.052 +10:00 [Verbose] Trying to load WIM metadata file at '"C:\install.wim"'
2019-06-25 18:49:54.069 +10:00 [Verbose] Getting WIM stream
2019-06-25 18:49:54.069 +10:00 [Verbose] (WIM) Finding Magic Bytes...
2019-06-25 18:49:54.070 +10:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-06-25 18:49:54.070 +10:00 [Verbose] (WIM) Finding WIM XML Data...
2019-06-25 18:49:54.070 +10:00 [Verbose] (WIM) Found WIM XML Data at 002
```

Geeviz, 2019-06-25:
I fixed it by installing a fresh Windows 10 (Pro, version may 2019) .. Now all works flawless :)

closed 2019-06-25 by Geeviz