WOA-Project/WOA-Deployer-Rpi #41: Raspberry pi 4 ?

squadchannel, 2019-06-24:
it can?

SteveALee, 2019-06-24:
My thoughts too. Just ordered a 2GB board so will be able to play soon

SteveALee, 2019-06-26:
Doesn't work for me. I followed the intructions to the letter using the latest Insiders Build (could not find 1903). The Pi never boots. Just flashes the green LED on a "4 flash - pause on" cycle.

* I have a 64GB card - so FAT32 formatted it first
* I could not eject the card from windows as it was in use but I was unable to find what was locking it.
* I'm not sure if all the partitions on the card are correct as one is empty - #43

SteveALee, 2019-06-26:
It seems [TOM's hardward failed to get it to boot](https://www.tomshardware.com/reviews/raspberry-pi-4-b,6193.html) but do not say how they deployed

squadchannel, 2019-06-26:
Just wait for support?

SuperJMN, 2019-06-26:
Sorry, guys. This isn't going to happen unless the OEM vendors, primarily Qualcomm, releases drivers for ARM64. We can make a campaign for that, but if we don't have drivers, we're sold.

SteveALee, 2019-06-26:
Ah, so it’s a different architecture? I hadn’t realised that.

-- 
Steve Lee
OpenDirective http://opendirective.com

SteveALee, 2019-06-27:
Looks like [some are available](https://discourse.pi64.win/t/status-of-the-arm64-device-drivers-as-of-12-feb-2019/38) already

SuperJMN, 2019-06-27:
Those drivers are for Raspberry Pi 3. If the 4 has different devices, it will require different drivers, too.

SuperJMN, 2019-06-27:
@SteveALee they are the same architecture: ARM64.

SteveALee, 2019-06-27:
@SuperJMN ah OK, I mis understood when you said "releases drivers for ARM64". Anyway I guess that table on pi64 will need a brand new column for pi 4 chips :)

Alex-Lewa-95-RPI-Noob, 2019-10-13:
Comment from SteveALee + my Comment.
> Doesn't work for me. I followed the intructions to the letter using the latest Insiders Build (could
>   not find 1903). The Pi never boots. Just flashes the green LED on a "4 flash - pause on" [...]

I have the same problem. + After the format of the SDcard with WOA the Windows disk management cant format the SDcard and cant open it as a folder, the ''MiniTool Partition Wizard'' can help you.
You need to create a exFAT Volume and then use the ''guiformat from ridgecrop'' to create a FAT32 partition.

> The Pi never boots. Just flashes the green LED on a "4 flash - pause on" cycle.

Is the SPI EEPROM corrupted?

>     * I could not eject the card from windows as it was in use but I was unable to find what was locking it.
You can shutdown your Computer and then eject the SDcard

hwangjr, 2019-11-13:
so, any progress for pi 4? guys.

linxu, 2019-12-03:
not support pi4?

squadchannel, 2019-12-03:
yep not supported. But I think they are working hard for support. Let's wait a little longer.

squadchannel, 2019-12-03:
Apparently the driver is not yet supported.

georgewoodall82, 2019-12-10:
Does it work on raspberry pi 3B+ or 3B?

Alex-Lewa-95-RPI-Noob, 2019-12-19:
> 
> 
> Does it work on raspberry pi 3B+ or 3B?

https://github.com/WOA-Project/WOA-Deployer-Rpi#requirements

Alex-Lewa-95-RPI-Noob, 2019-12-19:
Did you see this?

Support for the Raspberry Pi 4
https://github.com/andreiw/RaspberryPiPkg/issues/130

lampone_build.sh
https://github.com/andreiw/lampone/blob/master/lampone_build.sh

Pi4 Firmware Task Force 
https://github.com/pftf

lseurttyuu, 2020-01-29:
Somebody managed to run Windows 10 ARM on RP44, here's the news:

https://www.windowslatest.com/2020/01/27/windows-10-arm-on-raspberry-pi-4/

CornyjK, 2020-01-30:
NO NOT OPEN ANOTHER DUPLICATE REQUESTS, WE CAN'T RIGHT NOW.

closed 2020-01-31 by squadchannel