WOA-Project/WOA-Deployer-Rpi #45: Crashed before finishing now I have two unusable SD cards

georgelimacamara, 2019-06-29:
Hey,

The sw crashed and I tried with two different SD cards, now I can´t use them anymore before it says it's write protected or the formatting program was not able to do it so.

SteveALee, 2019-06-30:
Did you try the official [sd formatter](https://www.sdcard.org/downloads/formatter)? It should also remove the partitions.

georgelimacamara, 2019-07-01:
Yes, I tried so many things, procedures, formatting and partition sws.
But till now I not able to clear Current Read-Only State.

oldieman, 2019-08-24:
I have the same problem and 2 SD cards which SD formatter is unable to format.
When debugging the source code, I get following exception:
Error: Zeilennummer "59" und Zeilenposition "35" von "Die Angabe eines Werts für "System.Windows.StaticResourceExtension" führte zu einer Ausnahme.".
File: MarkdownViewerWindow.xaml

CornyjK, 2019-11-14:
Just make diskpart clean and you are fine.

closed 2019-11-14 by CornyjK