WOA-Project/WOA-Deployer-Rpi #47: SDCard not showing up

goisneto, 2019-08-11:
Im trying to install windows 10 Pro Build 18956, but in GUI my sdcard not showing up, and in cmd line `"C:\Users\Administrador\Downloads\WOA.Deployer\WoaDeployer.exe" deploy --disk 2 --wim install.wim` give me some erros 

**Screenshots**
My SDCARD showing normaly  on diskmgmt.msc
[image]
and on explorer
[image]
Using GUI utility
[image]
Using cmd line utility
[image]

**Log file**
GUI logs:
[Log-20190811.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/3489494/Log-20190811.txt)
CMD Logs:
```
C:\Users\Administrador\Downloads\WOA.Deployer>
[00:55:42 INF] Downloading UEFI
[00:55:42 WRN] UEFI was already downloaded. Skipping download.
[00:55:42 INF] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan95xx-windows/lan9500-wdf-v18.12.18.0.zip
[00:55:42 INF] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan78xx-windows/lan7800-wdf-v18.12.14.0.zip
[00:55:42 INF] Fetching zip from https://pi64.win/wp-content/uploads/2019/02/usb-drivers.zip
[00:55:42 INF] Fetching from GitHub subfolder: https://github.com/driver1998/bsp/bsp-master/prebuilt to Downloaded\Drivers\BSP Drivers
[00:55:42 WRN] https://github.com/driver1998/bspbsp-master/prebuilt was already downloaded. Skipping download.
[00:55:42 INF] License from Downloaded\Drivers\USB\license.md
By continuing you are accepting the following license below.
If you decline it, press Control+C anytime during the deployment process.
# TrueTask® USB
## ARM64 host drivers for Windows 10 on the Raspberry Pi
### Software License Agreement and Warranty Statement

MCCI IS WILLING TO LICENSE THE SOFTWARE TO YOU ONLY UPON THE CONDITION THAT YOU ACCEPT ALL OF THE TERMS CONTAINED IN THIS LICENSE AGREEMENT.  PLEASE READ THE TERMS CAREFULLY AND CLICK ON "ACCEPT" BEFORE INSTALLING THE SOFTWARE, AS CLICKING ON "ACCEPT" AND INSTALLING THE SOFTWARE WILL INDICATE YOUR AGREEMENT WITH THEM.  IF YOU DO NOT AGREE WITH THESE TERMS, THEN MCCI IS UNWILLING TO LICENSE THE SOFTWARE TO YOU, IN WHICH EVENT YOU SHOULD NOT PROCEED WITH INSTALLING THE SOFTWARE.

TO THE EXTENT THESE TERMS CONFLICT WITH ANY PREVIOUSLY SIGNED AND WRITTEN AGREED TO TERMS BETWEEN YOU AND MCCI, THE PRIOR TERMS SHALL CONTROL.

The software which accompanies this license (the "Software") and any accompanying documentation (the "Documentation") is the property of MCCI Corporation ("MCCI") or its licensors and is protected by copyright law. This agreement is between MCCI and the individual who downloads and wishes to use the Software and Documentation ("You"). While MCCI continues to own the software, You will have certain rights to use the Software upon your acceptance of this license.  MCCI grants You a non-transferable and non-exclusive license to use this Software under the terms of this agreement.  MCCI remains the proprietor of this software and licenses its use to You.  You do not obtain title to the Software or Documentation or any copyrights or proprietary rights in the software.  You assume responsibility for the selection of the Software to achieve your intended results, and for the installation, use and results obtained from the Software.  Additional rights and obligations regarding the Software and its contents, and/or the Documentation may be defined by a separate written agreement with MCCI, and if so, such separate written agreement shall be controlling.

In the absence of conflict of such separate written agreement or except as may be modified by such a license addendum which accompanies this license, your rights and obligations with respect to use of this Software and Documentation are as follows:

This license does NOT extend to any corporation or organization of which You are a member or with which You are affiliated. It is only for personal use. Commercial licenses for corporations and organizations are available from MCCI.

YOU MAY:  License Grant: You are granted non-exclusive rights to install and use the Software for personal use and evaluation purposes only. You may install on any Raspberry Pi computer that you personally own, provided that You acquire and dedicate a licensed copy of the Software for each computer on which the Software is used or to which it is transmitted over the internal network. You may also make backup copies of the Software.

RESTRICTIONS  YOU MAY NOT: (i) permit others to use the Software, except as expressly provided above for authorized network use; (ii) modify or translate the Software; (iii) reverse engineer, decompile, or disassemble the Software, except to the extent this restriction is expressly prohibited by applicable law; (iv) create derivative works based on the Software; (v) merge the Software with another product; (vi) export or use the Software data compilations, structures, or algorithms with another product; (vii) copy the Software, except as expressly provided above; (viii) remove or obscure any proprietary rights notices or labels on the Software; (ix) post the software on a website for public download, or (x) resell or distribute the Software, either stand-alone or bundled with or installed with hardware or software supplied by You or others, or (xi) distribute the Software or Documentation in any form (electronic or otherwise).

TERM AND TERMINATION.  The license provided in this Agreement will continue in perpetuity unless You fail to comply with the terms and conditions of this Agreement.  You agree that, upon such termination, you will either destroy (or permanently erase) all copies of the Software and Documentation, or return the original Software and Documentation to MCCI, together with any other material you have received from MCCI in connection with the Software.

TRANSFERS.  You may not transfer the Software or any rights under this Agreement without the prior written consent of MCCI. Any attempted transfer or assignment in violation of this provision shall be null and void.

FEEDBACK.  You agree that in the event You voluntarily disclose any ideas or suggestions to MCCI (in any manner, whether in writing or orally or otherwise) regarding the Software, Documentation, or Design Techniques, including possible enhancements or improvements ("Feedback"), MCCI may freely use and disseminate such Feedback.  You agree not to claim that MCCI owes You any compensation for its use or dissemination of such Feedback.

OWNERSHIP.  MCCI and its suppliers own the Software and all intellectual property rights embodied therein, including copyrights and valuable trade secrets embodied in the Software&#39;s design and coding methodology.  The Software is protected by United States copyright laws and international treaty provisions.  This Agreement provides You only a limited use license, and no ownership of any intellectual property. All content accessed through the Software is the property of the applicable content owner and may be protected by applicable copyright law. This license gives You no rights to such content.

WRITTEN RECORD.  The text of this Agreement is included with the Software files as "ttusb-pi64-installer-license.rtf."  You agree to print this text file immediately after installation of the Software and to maintain the printed copy as a written record of this transaction.

DISCLAIMER OF WARRANTY; LIMITATION OF LIABILITY **.**  MCCI PROVIDES THE SOFTWARE AND THE DOCUMENTATION "AS IS" WITHOUT WARRANTY OF ANY KIND EITHER EXPRESS IMPLIED OR STATUTORY, INCLUDING WARRANTIES OF TITLE, NON-INFRINGEMENT, MERCHANTABILITY, AND FITNESS FOR A PARTICULAR PURPOSE.  THERE IS NO WARRANTY OR GUARANTEE THAT THE OPERATION OF THE SOFTWARE WILL BE UNINTERRUPTED, ERROR-FREE, OR VIRUS-FREE, OR THAT THE SOFTWARE WILL MEET ANY PARTICULAR CRITERIA OF PERFORMANCE OR QUALITY EXCEPT AS EXPRESSLY PROVIDED IN THE LIMITED WARRANTY.  All risk of quality and performance of the software and documentation is with You.   This disclaimer of warranty constitutes an essential part of this Agreement.

To the extent that this Warranty Statement is inconsistent with the jurisdiction where You use the Software, the Warranty Statement shall be deemed to be modified consistent with such local law but to the maximum extent enforceable in such jurisdiction.  Under such local law, certain limitations may not apply, and You may have additional rights which vary from jurisdiction to jurisdiction.  For example, some states in the United States and some jurisdictions outside the United States may: (i) preclude the disclaimers and limitations of this Warranty Statement fr [... cut ...]
   em MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   em MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   em System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   em System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   em System.Windows.Application.RunDispatcher(Object ignore)
   em System.Windows.Application.RunInternal(Window window)
   em System.Windows.Application.Run(Window window)
   em Deployer.Raspberry.Gui.App.Main()
```

SuperJMN, 2019-12-30:
Does this happen with the latest version?

Biswa96, 2020-01-10:
Same situation. With latest version of WOA, the following error is shown in administrative command prompt and the GUI is same as OP.

<details>
<summary>Click to open</summary>

```cs
[14:42:43 INF] Downloading UEFI
[14:43:02 INF] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan95xx-windows/lan9500-wdf-v18.12.18.0.zip
[14:43:09 INF] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan78xx-windows/lan7800-wdf-v18.12.14.0.zip
[14:43:13 INF] Fetching zip from https://pi64.win/wp-content/uploads/2019/02/usb-drivers.zip
[14:43:26 INF] Fetching from GitHub subfolder: https://github.com/driver1998/bsp/bsp-master/prebuilt to Downloaded\Drivers\BSP Drivers
[14:43:27 INF] License from Downloaded\Drivers\USB\license.md
By continuing you are accepting the following license below.
If you decline it, press Control+C anytime during the deployment process.
# TrueTaskr USB
## ARM64 host drivers for Windows 10 on the Raspberry Pi
### Software License Agreement and Warranty Statement

License here... (Redacted)

[14:43:27 INF] Deploying Windows
[14:43:31 FTL] Operation failed
System.InvalidOperationException: Sequence contains no matching element
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   at Deployer.Raspberry.RaspberryPi.<GetDeviceDisk>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Tasks.DeployWindows.<Execute>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.WoaDeployer.<Deploy>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.Console.Program.<Execute>d__1.MoveNext() in D:\a\1\s\Source\Deployer.Raspberry.Console\Program.cs:line 49
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.Console.Program.<Main>d__0.MoveNext() in D:\a\1\s\Source\Deployer.Raspberry.Console\Program.cs:line 28

Unhandled Exception: System.AggregateException: One or more errors occurred. ---> System.InvalidOperationException: Sequence contains no matching element
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   at Deployer.Raspberry.RaspberryPi.<GetDeviceDisk>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Tasks.DeployWindows.<Execute>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Raspberry.WoaDeployer.<Deploy>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
[... 18 lines cut ...]
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(Object state)
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
[... cut ...]
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at System.Windows.Application.Run(Window window)
   at Deployer.Raspberry.Gui.App.Main()
```

</details>