WOA-Project/WOA-Deployer-Rpi #48: Installer OS

minecraftchest1, 2019-08-24:
I like this project. I am still trying to get the pi to boot to windows, but that is because I had to do it a hard way. Anyways, would it be to hard to create an OS image that would install the OS to the SD card? Currently, I am having to copy the OS files over from an VHD file to my USB drive, from a VM, because I do not have the correct permissions on any of or PCs. I would do this myself, but I do not know enough about what is needed for it to work to do it myself. Please respond when able.

minecraftchest1, 2019-08-24:
Should be a feature request.

SuperJMN, 2019-12-30:
I'm sorry, but to install WOA you definitely need to do some special steps. Just copying files won't make WOA boot. That's why I created WOA Deployer, to make things easier.

closed 2019-12-30 by SuperJMN