WOA-Project/WOA-Deployer-Rpi #59: It always quit suddenly after browsing the file "install.wim"

MZW-BG, 2020-01-31:
I'm using an Windows To Go system, of which the system version is Windows 10 version 1903, 18362.418, ~~because I have only one computer now(except the unusable RPi), and I can't go out for other computers due to the virus 2019-nCoV,~~ and, when I was using my origin Windows 7, the WOA **always** stuck at loading the drivers.

**To Reproduce**
I just open the WoaDeployer.exe, choose the SD Card, then find my "install.wim", then it stucked, after seconds it quit without any message.
The "install.wim" is from "cn_windows_10_business_edition_version_1809_updated_sept_2018_arm64_dvd_104983b1"

**Log file**
[Log-20200131.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/4139416/Log-20200131.txt)
I think the XML file which is thought wrong is below:
[[1].xml.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/4139454/1.xml.txt)
At the root directory.

I have compared the SHA1 and made sure the ISO file is correct.

This tool is pretty cool and useful, I hope it can work soon!

SuperJMN, 2020-01-31:
Please, use another WIM file. Some users have reported invalid WIMs before. It's not a problem with the Deployer, but the file having and invalid metadata.

Thanks for your kind words! If you keep having problems, write me here 😊

CornyjK, 2020-01-31:
Note that deploying from Windows 7 is not supported and deployment from WTG is untested and will never be supported as Microsoft deprecated this feature.

MZW-BG, 2020-01-31:
> Please, use another WIM file. Some users have reported invalid WIMs before. It's not a problem with the Deployer, but the file having and invalid metadata. Thanks for your kind words! If you keep having problems, write me here 😊
> […](#)

Can the .wim files from non-arm64 versions be used? I just cannot find another ISO file!

SuperJMN, 2020-01-31:
No, just ARM64 versions. You can generate your own WIM files using this site: http://uupdump.ml

MZW-BG, 2020-02-01:
Err... Another problem appeared.
I get a usable install.wim but when deploying, it said "Error: The read operation failed, see inner exception."
I noticed that the problem appeared just after "Downloading UEFI".
May be a problem of my devices?
Log files are below:

[Log-20200201-upload.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/4142074/Log-20200201-upload.txt)

MZW-BG, 2020-02-01:
I've checked out that my SD Card is not write-protected, and it is formatted with FAT32.

MZW-BG, 2020-02-01:
Another question, can this thing work on Rpi4?
It's said that it can, but... I ain't sure.

boram4620, 2020-02-01:
I am working on my RPI4 now.. and having the same issue. 
As SuperJMN said, I am gonna try with another wim file. wish It would work well..
I will keep you posted. Good luck!

CornyjK, 2020-02-01:
Do not hijack threads, use existing issue threads or create new issues.

MZW-BG, 2020-02-01:
OK, so this problem is solved.

closed 2020-02-01 by MZW-BG