WOA-Project/WOA-Deployer-Rpi #60: Error: The read operation failed, see inner exception.

MZW-BG, 2020-02-01:
Err... Another problem appeared.
I get a usable install.wim but when deploying, it said "Error: The read operation failed, see inner exception."
I noticed that the problem appeared just after "Downloading UEFI".
May be a problem of my devices?
Log files are below:
[Log-20200201-upload.txt](https://github.com/WOA-Project/WOA-Deployer-Rpi/files/4142074/Log-20200201-upload.txt)

MZW-BG, 2020-02-01:
I've checked out that my SD Card is not write-protected, and it is formatted with FAT32.

MZW-BG, 2020-02-02:
Nobody here?

SuperJMN, 2020-02-02:
Can you please try another card?

MZW-BG, 2020-02-02:
I have only one SD Card now and I cannot go out.... I am trying for one more time.

MZW-BG, 2020-02-02:
OK... It might be my fault... It seems to work now.Thank you all the same!

closed 2020-02-02 by MZW-BG