WOA-Project/WOA-Deployer-Rpi #64: rpi4 win10 17134

avenir-1023, 2020-02-13:
Developer, hello. 
Recently I saw a successful implementation of win10arm on raspberry PI 4B on the Internet. Win10 was successfully started using win10 17134. But I couldn't replicate it on the raspberry pi. Excuse me developer, recently can you study raspberry PI 4B win10 successful operation method or tutorial?

Marcinoo97, 2020-02-16:
https://discourse.pi64.win/t/how-to-installing-windows-10-arm64-17134-on-your-raspberry-pi-3-and-4/488

dudu4, 2020-02-29:
hey marcinoo i don't have raspberry pi 3. can you send me a backup of my USB drive. can you give me a little help with the file I lost my mega account if you can send the file to my email

bzy-080408, 2020-05-04:
Here's my approach:
1. Use WOA to select the language, device, mirror, drive to use the latest version on the server, and then partition the table to select MBR mode, and then brush.
2. Use the windows script I made, link:
https://github.com/bzy-080408/raspberrypi4-windows10/releases/tag/v1.1.5 If there's any thing You can find, I'll correct it.

Marcinoo97, 2020-05-04:
More people are comfortable with runnign a GUI than a script so on https://rpi4-uefi.dev/win10-arm64-on-pi-4b/ there is a guide how to use WoR 2.0.0-alpha1 to install windows 10 on raspberry pi 4. There is no reason to develop another tool to deploy windows 10 to raspberry pi 4. We already have WoR and soon WOA deployer will be updated. You can help to update chinese translation of WoR if you can.  6 lines are in english instead of chinese here https://github.com/worproject/WoR-Translations/blob/v2.0.0-alpha.1/lang/zh-CN.lng

bzy-080408, 2020-05-05:
However, The RASP4 USB PORT IS NOT AVAILABLE WHEN I'M USING IT

bzy-080408, 2020-05-27:
And I want to know why when I run ubuntu20.10(uefi install) I can use the USB but when I run win10 on it I can`t use?

CharlotteDunois, 2020-06-12:
@bzy-080408 According to the guide only the USB-C works correctly. The USB2 and USB3 Type A ports don't work. WiFi is not supported yet, at least during setup. I have yet to find out whether it will work later on.

I can confirm that with Windows 10 Pro Build 19640 on my RPi 4B 8G.

bzy-080408, 2020-06-27:
@CharlotteDunois but why nobody to build a USB drive, The source code may be is here: https://github.com/tianocore/edk2-platforms/tree/master/Drivers/OptionRomPkg/Bus/Usb/FtdiUsbSerialDxe

Artells, 2020-07-02:
> 这是我的方法：
> 
> 1. 使用WOA选择语言，设备，镜像，驱动器以使用服务器上的最新版本，然后对表进行分区以选择MBR模式，然后进行刷。
> 2. 使用我制作的Windows脚本，链接：
>    [https](https://github.com/bzy-080408/raspberrypi4-windows10/releases/tag/v1.1.5) : [//github.com/bzy-080408/raspberrypi4-windows10/releases/tag/v1.1.5](https://github.com/bzy-080408/raspberrypi4-windows10/releases/tag/v1.1.5)如果您可以找到任何东西，我会予以纠正。

The latest progress, the raspberry pie can be turned on, but running memory can only be set to 1G, and OOBE will then be blue screened, with no solution for the time being.

bzy-080408, 2020-07-02:
@Artells 来自中国？come from China?

dudu4, 2020-07-15:
how to apply the drivers for USB type A on the raspberry pi 4 does anyone know how to apply correctly if you can show me how to do it
appreciate

MMan1982, 2020-07-16:
Is anyone know a working built-in ethernet for pi4?