WOA-Project/WOA-Deployer-Rpi #72: "Downloaded\Drivers\BSP Drivers" not found

Leviathan09, 2020-07-30:
**Describe the bug**
Fetching from GitHub subfolder: https://github.com/driver1998/bsp/bsp-master/prebuilt to Downloaded\Drivers\BSP Drivers is not possible because the link is invalid
The part "/bsp-master/prebuilt" doesn't exist and manually i also couldn't found any prebuilt drivers

**To Reproduce**
Start WOA
Go through alle steps and click deploy
At the end of the deploy process it says DISM Error 2
If i look into the logs i can see that the error tells me "Downloaded\Drivers\BSP Drivers" couldn't be found

**Expected behavior**
Download the correct drivers and deploy it

**Log file**
I don't have it in english, just german(sry)

```
Starting process { FileName: "C:\WINDOWS\SysNative\dism.exe", Arguments: "/Add-Driver /Image:F:\ /Driver:\"Downloaded\Drivers\BSP Drivers\" /Recurse /ForceUnsigned" }
2020-07-30 00:52:01.411 +02:00 [Verbose] Process started sucessfully
2020-07-30 00:52:01.433 +02:00 [Verbose] 
2020-07-30 00:52:01.433 +02:00 [Verbose] Tool zur Imageverwaltung fr die Bereitstellung
2020-07-30 00:52:01.433 +02:00 [Verbose] Version: 10.0.19041.329
2020-07-30 00:52:01.433 +02:00 [Verbose] 
2020-07-30 00:52:10.564 +02:00 [Verbose] Abbildversion: 10.0.20161.1000
2020-07-30 00:52:10.564 +02:00 [Verbose] 
2020-07-30 00:52:12.765 +02:00 [Verbose] 
2020-07-30 00:52:12.765 +02:00 [Verbose] Fehler: 2
2020-07-30 00:52:12.765 +02:00 [Verbose] 
2020-07-30 00:52:12.765 +02:00 [Verbose] Die angegebene Datei "Downloaded\Drivers\BSP Drivers" kann nicht gefunden werden.
2020-07-30 00:52:12.765 +02:00 [Verbose] 
```

Melonenyoshi, 2020-07-30:
same for me and i dont know how to fix this

Melonenyoshi, 2020-07-30:
and i looked in the explorer that file is not there so it isnt a bug in the source code

Leviathan09, 2020-07-30:
I will try to create the folder by my self and copy these drivers there (https://github.com/worproject/RPi-Windows-Drivers/releases)
Maybe it will work

Flihx, 2020-07-30:
I get the same error - Did it work?

Leviathan09, 2020-07-30:
Nope, didn't work for me. Neither with USB-Stick or SD-Card.
Flashed now a SD-Card with "https://www.worproject.ml/downloads" Version 1.5.2
And with the following Win 10 ARM Image "https://uupdump.ml/selectlang.php?id=878910a0-bef6-4449-b2da-591c2b58d847"

Now the Pi booted up to Win 10 but it's very slow, actually slower than slow.

Leviathan09, 2020-07-30:
> Nope, didn't work for me. Neither with USB-Stick or SD-Card.
> Flashed now a SD-Card with "https://www.worproject.ml/downloads" Version 1.5.2
> And with the following Win 10 ARM Image [...]

Flashed the same image with WoR 1.5.2 to a USB Stick
It's working and the Pi 3B+ is booting
Also it's a little bit faster then the SD card

Flihx, 2020-07-30:
It actually worked for me. I took the "RPi4_Windows_ARM64_Drivers_v0.1.zip", unzipped it and then created a Folder called Drivers in Downloaded. I then made a folder called "BSP Drivers" and took all the Subfolders from the Zip and but them in the BSP Drivers folder. Then it worked. Glad it works anyway tho.

Leviathan09, 2020-07-30:
> 
> 
> It actually worked for me. I took the "RPi4_Windows_ARM64_Drivers_v0.1.zip", unzipped it and then created a Folder called Drivers in Downloaded. I then made a folder called "BSP Drivers" and took all the Subfolders from the Zip and put them in the BSP Drivers folder. Then it worked. Glad it works anyway tho.

Which Win 10 ARM Image did you use?

Flihx, 2020-07-30:
I haven't tried starting the PI actually, i just referred to the Installer to work. I'm using the 19645 Windows Pro Image

Flihx, 2020-07-30:
I have tried starting the Pi, and the Flashing green light indicates, that the image is corrupted. I assume there is a problem with the Image tho. I'm going to try another one and report afterwards!

Leviathan09, 2020-07-30:
Tried your was with the drivers again.
Flashing worked with a 20xxx Image.
But it only Starts in the UEFI Bootmenü

paulstelian97, 2020-09-11:
Also affected and the manual fix (download fix, extract to new folder "BSP Drivers") seems to work.

robertkross, 2020-10-13:
Managed to flash the SD card, but now I am having the same problem as @Flihx. Doesn't boot, green light flashes 4 times. Did anyone manage to solve this problem?

NikoHD203, 2020-12-06:
I've tried to download the BSP Drivers by myself ([link](https://github.com/driver1998/bsp/releases/download/v0.5.2/rpi-arm64-bsp.zip)) and extracted the content into Downloaded\Drivers\BSP Drivers. There was no error, but the RPi didn't boot.