WOA-Project/WOA-Deployer-Rpi #76: problem during deployment DISM code 21.

mauriciosena, 2020-11-16:
`2020-11-16 16:48:28.328 -03:00 [Verbose] The '"F:\sources"' does not exist
2020-11-16 16:48:33.219 -03:00 [Verbose] Default directory for WimFolder has been set to "D:\sources"
2020-11-16 16:48:33.229 -03:00 [Verbose] Trying to load WIM metadata file at '"D:\sources\install.wim"'
2020-11-16 16:48:33.295 -03:00 [Verbose] Getting WIM stream
2020-11-16 16:48:33.298 -03:00 [Verbose] (WIM) Finding Magic Bytes...
2020-11-16 16:48:33.300 -03:00 [Verbose] (WIM) Found Magic Bytes at 0
2020-11-16 16:48:33.301 -03:00 [Verbose] (WIM) Finding WIM XML Data...
2020-11-16 16:48:33.301 -03:00 [Verbose] (WIM) Found WIM XML Data at 044478675542
2020-11-16 16:48:33.358 -03:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4447867554", Images: [ImageMetadata { DirectoryCount: "24062", FileCount: "98114", TotalBytes: "16976807768", HardLinkBytes: "6424763358", CreationTime: Time { HighPart: "0x01D5AD23", LowPart: "0xA3F93EF6" }, LastModificationTime: Time { HighPart: "0x01D6BC4E", LowPart: "0x72E00B08" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Sistema Operacional Microsoft® Windows®", EditionId: "Professional", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.19041.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "pt-BR", Default: "pt-BR" }, Version: Version { Major: "10", Minor: "0", Build: "19041", SpBuild: "330", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Pro", Description: "Windows 10 Pro", Flags: null, DiplayName: "Windows 10 Pro", DisplayDescription: "Windows 10 Pro", Index: "1" }] }
2020-11-16 16:48:33.366 -03:00 [Verbose] WIM metadata file at '"D:\sources\install.wim"' retrieved correctly
2020-11-16 16:48:39.858 -03:00 [Information] Downloading UEFI
2020-11-16 16:48:39.866 -03:00 [Warning] UEFI was already downloaded. Skipping download.
2020-11-16 16:48:39.879 -03:00 [Information] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan95xx-windows/lan9500-wdf-v18.12.18.0.zip
2020-11-16 16:48:39.897 -03:00 [Information] Fetching zip from http://ww1.microchip.com/downloads/en//softwarelibrary/obj-lan78xx-windows/lan7800-wdf-v18.12.14.0.zip
2020-11-16 16:48:39.909 -03:00 [Information] Fetching zip from https://pi64.win/wp-content/uploads/2019/02/usb-drivers.zip
2020-11-16 16:48:39.921 -03:00 [Information] Fetching from GitHub subfolder: https://github.com/driver1998/bsp/bsp-master/prebuilt to Downloaded\Drivers\BSP Drivers
2020-11-16 16:48:43.505 -03:00 [Information] License from Downloaded\Drivers\USB\license.md
2020-11-16 16:48:45.400 -03:00 [Information] Deploying Windows
2020-11-16 16:48:45.408 -03:00 [Information] Flashing GPT image...
2020-11-16 16:48:45.415 -03:00 [Verbose] We are about to run Etcher: "Core\Tools\Etcher-Cli\x86\Etcher" "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount"
2020-11-16 16:48:45.418 -03:00 [Verbose] Starting process { FileName: "Core\Tools\Etcher-Cli\x86\Etcher", Arguments: "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount" }
2020-11-16 16:48:45.441 -03:00 [Verbose] Process started sucessfully
2020-11-16 16:49:26.854 -03:00 [Information] Validating flashed image...
2020-11-16 16:50:01.305 -03:00 [Information] GPT image flashed
2020-11-16 16:50:01.311 -03:00 [Verbose] Creating Windows partition...
2020-11-16 16:50:03.268 -03:00 [Verbose] Mounting volume "Label: , Size: 0 b, Partition: Disk: Number: 1, Size: 14,84 GB, AllocatedSize: 14,84 GB, Number: 3, Letter: "
2020-11-16 16:50:10.328 -03:00 [Verbose] Windows Partition created successfully
2020-11-16 16:50:10.332 -03:00 [Information] Preparing for Windows deployment
2020-11-16 16:50:10.335 -03:00 [Verbose] Getting "WindowsARM" volume
2020-11-16 16:50:11.242 -03:00 [Information] Deploying Windows...
2020-11-16 16:50:11.247 -03:00 [Verbose] Checking image at "D:\sources\install.wim", with index "D:\sources\install.wim"
2020-11-16 16:50:11.247 -03:00 [Verbose] Image file at '"D:\sources\install.wim"' exists
2020-11-16 16:50:11.247 -03:00 [Verbose] Getting WIM stream
2020-11-16 16:50:11.247 -03:00 [Verbose] (WIM) Finding Magic Bytes...
2020-11-16 16:50:11.247 -03:00 [Verbose] (WIM) Found Magic Bytes at 0
2020-11-16 16:50:11.247 -03:00 [Verbose] (WIM) Finding WIM XML Data...
2020-11-16 16:50:11.248 -03:00 [Verbose] (WIM) Found WIM XML Data at 044478675542
2020-11-16 16:50:11.266 -03:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4447867554", Images: [ImageMetadata { DirectoryCount: "24062", FileCount: "98114", TotalBytes: "16976807768", HardLinkBytes: "6424763358", CreationTime: Time { HighPart: "0x01D5AD23", LowPart: "0xA3F93EF6" }, LastModificationTime: Time { HighPart: "0x01D6BC4E", LowPart: "0x72E00B08" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Sistema Operacional Microsoft® Windows®", EditionId: "Professional", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.19041.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "pt-BR", Default: "pt-BR" }, Version: Version { Major: "10", Minor: "0", Build: "19041", SpBuild: "330", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Pro", Description: "Windows 10 Pro", Flags: null, DiplayName: "Windows 10 Pro", DisplayDescription: "Windows 10 Pro", Index: "1" }] }
2020-11-16 16:50:11.266 -03:00 [Verbose] We are about to run DISM: "C:\WINDOWS\SysNative\dism.exe" "/Apply-Image  /ImageFile:\"D:\sources\install.wim\" /Index:1 /ApplyDir:F:\"
2020-11-16 16:50:11.266 -03:00 [Verbose] Starting process { FileName: "C:\WINDOWS\SysNative\dism.exe", Arguments: "/Apply-Image  /ImageFile:\"D:\sources\install.wim\" /Index:1 /ApplyDir:F:\" }
2020-11-16 16:50:11.277 -03:00 [Verbose] Process started sucessfully
2020-11-16 16:52:52.500 -03:00 [Error] An error has occurred
Deployer.Exceptions.DeploymentException: There has been a problem during deployment: DISM exited with code 21.
   em Deployer.NetFx.DismImageService.<ApplyImage>d__1.MoveNext() na D:\a\1\s\Source\DeployerPlatform\Deployer.NetFx\DismImageService.cs:linha 46
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.WindowsDeployer.<Deploy>d__7.MoveNext()
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.Tasks.DeployWindows.<Execute>d__4.MoveNext()
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.Raspberry.WoaDeployer.<Deploy>d__3.MoveNext()
--- Fim do rastreamento de pilha do local anterior onde a exceção foi gerada ---
   em System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   em System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   em Deployer.Raspberry.Gui.ViewModels.DeploymentViewModel.<Deploy>d__20.MoveNext() na D:\a\1\s\Source\Deployer.Raspberry.Gui\ViewModels\DeploymentViewModel.cs:linha 82
2020-11-16 16:52:52.534 -03:00 [Information] Error: There has been a problem during deployment: DISM exited with code 21.
2020-11-16 17:01:49.845 -03:00 [Information] Downloading UEFI
[... cut ...]
2020-11-16 17:01:53.914 -03:00 [Information] Flashing GPT image...
2020-11-16 17:01:53.914 -03:00 [Verbose] We are about to run Etcher: "Core\Tools\Etcher-Cli\x86\Etcher" "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount"
2020-11-16 17:01:53.915 -03:00 [Verbose] Starting process { FileName: "Core\Tools\Etcher-Cli\x86\Etcher", Arguments: "-d \\.\PHYSICALDRIVE1 \"Core\gpt.zip\" --yes --no-unmount" }
2020-11-16 17:01:53.976 -03:00 [Verbose] Process started sucessfully
`

Is there someone to help me on this?

jvoeller0129, 2021-05-03:
ok this may or may not help someone as i had the same issue... 
in the WOA folder theres a folder called scripts with deployment.txt
i removed from the file
FetchGitHubFolder "https://github.com/driver1998/bsp" "bsp-master/prebuilt" "Downloaded\Drivers\BSP Drivers"
ShowLicense "Downloaded\Drivers\USB\license.md"

manually downloaded the bsp drivers from 
https://github.com/worproject/RPi-Windows-Drivers/releases

made a folder in the downloaded\drivers called BSP Drivers
and unzipped the downloaded files to that see if this helps anyone

jvoeller0129, 2021-05-03:
This should fix it for RPI

1. Open WOA install folder
2. Open The Scripts folder
3. Edit Deployment.txt
4. Right above where it says show license remove that line delete the line
5. open https://github.com/worproject/RPi-Windows-Drivers/releases and right click the apporiate file for your platform and copy link address
6. back in deployment.txt above show License
7. type Fetch then paste the link and type "Downloaded\Drivers\BSP Drivers"
`Fetch "https://github.com/worproject/RPi-Windows-Drivers/releases/download/v0.9/RPi3_Windows_ARM64_Drivers_v0.9.zip" "Downloaded\Drivers\BSP Drivers"`
save file and redeploy

make sure the quotes are before https and at the end of .zip as well as before Downloaded and after Drivers